jingoro2112/wrench #1: Compile error on both Windows and Linux

eddavis2, 2022-08-26:
I tried building on both Windows and Linux.
On Both systems, I get the same error:

g++  -MD -Wall -Werror -O3  -I. -c -o objs_linux/cc.o discrete_src/cc.cpp
discrete_src/cc.cpp:2178:1: fatal error: opening dependency file objs_linux/cc.d: No such file or directory
 2178 | }
      | ^
compilation terminated.
make: *** [Makefile:43: objs_linux/cc.o] Error 1

Thanks for any help!

jingoro2112, 2022-09-09:
This happens if you don't run "make clean" first to create the directory, I'll fix it!

closed 2022-09-09 by jingoro2112

Diegovsky, 2022-10-07:
Still happens in 8c0447b

jingoro2112, 2022-10-08:
I added a blank file to the obj/ directory that should prevent it from
being omitted, should fix it going forward