jingoro2112/wrench #4: Segfault instead of compilation error

Diegovsky, 2022-10-07:
Here's a snippet that causes a segfault using the wrench cli with `wrench r <scriptname>`:
```javascript
function print(arg) {
    log(arg == null)
    log(arg);
}

print("Hello, world!");
```

I wrote it accidentaly missing the semicolon but got a segfault instead of a compilation error. I tried debugging it with GDB and apparently it is something to do with `wr_callFunction` being called with zeroed `argv` and `argn`.

jingoro2112, 2022-10-07:
well.. don't forget the semicolon! :)

I'll look into this right now

Diegovsky, 2022-10-07:
Won't forget again :)

jingoro2112, 2022-10-07:
That was a fun one. finally nailed it by comparing the expression resolves
actually completed to the list presented to the resolved. In the error case
they mismatch because the operators are not where they should be for a
valid expression.

fixed in latest, and thanks for the bugrep.

closed 2022-10-07 by jingoro2112

Diegovsky, 2022-10-08:
No problem! I'm glad to help :+1:

jingoro2112, 2022-10-11:
one thing though, there is no intrinsic "print" function.. I guess I should
add it to the default wrench cli. still it should just do nothing, not
segfault

commit 561f8a76 (2022-10-07): fixed #4 malformed code caused segfault

#4 fixed, wrench now properly recognizes this as a parse error where it didn't before:
 function print(arg) {
    log(arg == null)                 // <--- no ';' !!
    log(arg);
}

Thanks to Diego Augusto for identifying this one
added 'print' as an intrinsic to the wrench CLI. 
--- log is now deprecated