jingoro2112/wrench #7: wr_makeCharArray

travisapple, 2023-02-23:
First off. Wrench is fantastic. Curt, this project has saved my ass and will be running on an ESP32 in a product in the agriculture industry.

However, I need to push a string into a Wrench function. wrench.h defines

void wr_makeCharArray( WRValue* val, const unsigned char* data, const int len );

but wrench.cpp doesn't show any code for it. wr_makeInt() and wr_makeFloat() are there but not this one.

jingoro2112, 2023-02-23:
Wow cool! yeah I'll fix this up!

travisapple, 2023-02-23:
Thank you!

FYI: I have a class of devices in the field powered by ESP32 chips (with wifi) that test for crop disease. These devices need to have their code updated ALL THE TIME and this code is extremely security sensitive. They're physically all over the place so I cannot flash them all manually.

The devices will now be able to 're-flash' themselves over the internet with new C++ code (Wrench) every day. Using public/private keys this code will be completely encrypted during transfer. It's perfect!

jingoro2112, 2023-02-24:
This has been amended and fixed, along with a couple of bugfixes and enhancements.

closed 2023-02-24 by jingoro2112