jingoro2112/wrench #9: Segmentation fault at the end of the function execution.

glebnovodran, 2023-05-29:
Hello,

A reduced test case for getting it crashed is:

```
function faulty_func() {
	t = 1.0;
	x = t * 4.0;  // <-- any arithmetic operation involving a literal
	// can be x = 1.0 + 2.0;
	z = x * t; // <-- any binary operation on two variables
	return z;
}

```
The function should  contain an arithmetic operation involving a numeric literal followed by a binary arithmetic operation on two variables.

Debugging it, I figured out that this line `z = x * t;` produces WR_ERR_bad_expression when the expression is being resolved.
It happens [here](https://github.com/jingoro2112/wrench/blob/3838502c04fba683b4c0741bf370b8ca46615e19/src/wrench.cpp#L5135) . `resolves != startedWith`  for some reason.

Ultimately, it crashes at the end of the function execution.

jingoro2112, 2023-05-30:
I will put some time into this asap to get to the bottom

jingoro2112, 2023-06-02:
This turned out to be an unclean state in the keyhole optimizer, which has been fixed and regressed.

Thank you so much for taking the time to report it!

closed 2023-06-02 by jingoro2112

glebnovodran, 2023-06-03:
Glad to help.