jingoro2112/wrench #12: Something strange with globals

Buterbrott, 2023-06-16:
Strange globals collision.

print output should be constant 20, but in this test it depends on b variable value.

`
WRState* w;
WRContext* context;

void print( WRState* w, const WRValue* argv, const int argn, WRValue& retVal, void* usr )
{
	char buf[1024];
	for( int i=0; i<argn; ++i )
	{
		printf( "%s", argv[i].asString(buf, 1024) );
	}
}

void millis( WRState* w, const WRValue* argv, const int argn, WRValue& retVal, void* usr )
{
	wr_makeInt(&retVal,100000);
}

const char* wrenchCode = " \
b = 0.0; \
a = 10; \
function init(){ \
b = millis()/200.0; \
} \
function test(i,j){ \
print(i*a); \
} \
";

int main() {

  
  WRState* w = wr_newState(90); // create the state

	wr_registerFunction( w, "print", print );
  wr_registerFunction( w, "millis", millis );

	unsigned char* outBytes; // compiled code is alloc'ed
	int outLen;

  char* errorMsg;

	int err = wr_compile( wrenchCode, strlen(wrenchCode), &outBytes, &outLen, errorMsg); 
	if ( err == 0 )
	{

    WRContext* context = wr_run( w, outBytes, outLen );
    wr_callFunction( w, context, "init",NULL,0);
    WRValue wrCoords[2];
    wr_makeInt(&wrCoords[0], 2);
    wr_makeInt(&wrCoords[1], 3);
    wr_callFunction( w, context, "test",wrCoords,2);
	}

  delete[] outBytes; // clean up 
	wr_destroyState( w );

	return 0;

}
`

jingoro2112, 2023-06-17:
Ack.. ill look into this asap

Buterbrott, 2023-07-15:
Swapped registers assignment in

```
CASE(GLBinaryMultiplication):
{
  register1 = globalSpace + *pc++;
  register0 = frameBase + *pc++;
  wr_MultiplyBinary[(register0->type<<2)|register1->type]( register0, register1, stackTop++ );
  CONTINUE;
}
```

to

```
register0 = frameBase + *pc++;
register1 = globalSpace + *pc++;
```

And my example was calculated correctly

jingoro2112, 2023-07-15:
Thank you so much for figuring this out! I had not been able to find the
time to look into it. Fix made and published.

jingoro2112, 2023-07-15:
Fixed courtesy of the original reporter for finding AND tracking down the poroblem.

closed 2023-07-15 by jingoro2112

Buterbrott, 2023-07-16:
I think the same problem in GL(LG)BinarySubtraction. And you can't just swap register lines in Division and Subtraction, because of operands order.

maybe it would be better something like this?

```
CASE(GLBinarySubtraction):
{
  register1 = frameBase + *pc++;
  register0 = globalSpace + *pc++;
  wr_SubtractBinary[(register0->type<<2)|register1->type]( register0, register1, stackTop++ );
  CONTINUE;
}

CASE(LGBinarySubtraction):
{
  register1 = globalSpace + *pc++;
  register0 = frameBase + *pc++;
  wr_SubtractBinary[(register0->type<<2)|register1->type]( register0, register1, stackTop++ );
  CONTINUE;
}

CASE(GLBinaryDivision):
{
  register1 = frameBase + *pc++;
  register0 = globalSpace + *pc++;
  wr_DivideBinary[(register0->type<<2)|register1->type]( register0, register1, stackTop++ );
  CONTINUE;
}

CASE(LGBinaryDivision):
{
  register1 = globalSpace + *pc++;
  register0 = frameBase + *pc++;
  wr_DivideBinary[(register0->type<<2)|register1->type]( register0, register1, stackTop++ );
  CONTINUE;
}
```