jingoro2112/wrench #14: Printing float

matsekberg, 2023-08-16:
Should this print "12.45":

`str::printf( "%f", 12.45 );`

It doesn't I get an empty string.

jingoro2112, 2023-08-16:
I have a very code-optimized custom implementation of sprintf(...), written by me some time ago for embedded systems that don't have unlimited memory.

I considered adding floating-point support but decided against it since the whole idea was to keep it super-small and compact.

If you want %f / %g support feel free to add it, as a matter of fact I can probably just add a compiler define that uses the standard sprintf() instead of the embedded version, I'll let you know.

jingoro2112, 2023-08-16:
oh I remember why I didn't do this.. it was too much trouble to try and get a vararg-ish implementation piggybacked on the system call which has a list of WRValues. In order to do float I had to write and actual float parser... hrm well that's still kinda on the table but will eat some space, I'll see if I can add it without too much trouble.

jingoro2112, 2023-08-16:
Added `#define WRENCH_FLOAT_SPRINTF` to the latest version (2.08) which defaults off. turn it ON and %f/%g will work fully as sprintf(...) uses them.

closed 2023-08-16 by jingoro2112

matsekberg, 2023-08-16:
Great!
And if I could ask for another thing…
Documentation on how to write my own library, how the arguments are passed and how to return results. But I could dive into the code and try to figure it out also (not loving c++, it might take a while 😅)

jingoro2112, 2023-08-17:
There are quite a few examples, std_* are all implemented as libraries. I
detailed some of the info here:
https://home.workshopfriends.com/wrench/www/#libcalls

If you have specific questions I might be able to help.