jingoro2112/wrench #15: concatenate string

rjjrbatarao, 2023-12-11:
im trying to concatenate string like

message = "hello ";
message += "world";

the output is - hello

how can i do this? is strcat possible in wrench?

jingoro2112, 2023-12-11:
Okay just uploaded the new str:: library calls + test code. as 3.1.0

"str += str" is not going to happen today,. but I'll take a hard look on adding it.

closed 2023-12-11 by jingoro2112

jingoro2112, 2023-12-11:
HA! I could not get this itch to go away in my head and suddenly hit upon a way to do it, which worked!

3.1.1 now allows code like

str + str

AND

str += str

To work!! Specifically strings and addition only, and at no overhead to normally running code :)

rjjrbatarao, 2023-12-19:
wow amazing, thank you for this great addition will be testing it now.