jingoro2112/wrench #18: Support of lazy loading

colesnicov, 2024-01-14:
Hello. Here '[reddit](https://www.reddit.com/r/ProgrammingLanguages/comments/xa3rk6/updates_to_wrench_an_embeddedsystem_interpreter/)', I read from VM is able to load the code on the background and run it. Is he able to load the code on the first time and eventually read the needed part of the code? Something like a swap? I would need to start a more scopeful program that does not fit, all, to the RAM and so it would be necessary to load it after sometime and eventually, according to it, to load the part of the program from the lower or higher addresses.

jingoro2112, 2024-01-14:
This sounds like you want to run partial programs because it does not entirely fit into RAM?

First reaction is that's okay because Wrench executes from ROM. It does not make a copy of the program if you pass it compiled bytecode, it runs it in place. This was a frustration I had with other scripting languages that insisted on stuffing code into RAM to run it.

If you mean you are loading the program from some outside resource and can't fit it all into RAM to run it, wrench was originally designed with this in mind but has since been abandoned. It was too hard to make work for too little gain no one seemed to need.

If you have a more specific question/requirement I'd love to offer better advice, but quite obviously english is not your first language so there is a barrier :(

jingoro2112, 2024-01-18:
Sorry I don't plan to implement lazy loading, there are several hurdles that are not worth the effort at this time