jingoro2112/wrench #19: Can't compile bytecode

colesnicov, 2024-01-14:
I tried deploying version 2.8 on _ESP8266 under ESP8266-RTOS-SDK_ and everything worked on 'first click'. But today I updated the repository to the current version, **3.2.3**, and it stopped working.
The first problem was the inability to detect the bit order:
```
/components/wrench/src/wrench.h:626:2: error: #error "**Endian-ness not detected!** Please contact curt.hartung@gmail.com so it can be added"
 #error "Endian-ness not detected! Please contact curt.hartung@gmail.com so it can be added"
  ^~~~~
```

I fixed it by commenting out the line [626](https://github.com/jingoro2112/wrench/blob/b299e61d6d7b962be65f5201ea3954269c1f7a65/src/wrench.h#L626C1-L626C1),
and inserted the following lines of code below this line:
```
#include <cstdlib>
#include <memory.h>
#include <stdio.h>
#include <fcntl.h>

#if defined(XTHAL_BIGENDIAN) && XTHAL_BIGENDIAN
#warning "WRENCH_BIG_ENDIAN"
#define WRENCH_BIG_ENDIAN
#endif

#if defined(XTHAL_LITTLEENDIAN) && XTHAL_LITTLEENDIAN
#warning "WRENCH_LITTLEENDIAN"
#define WRENCH_LITTLE_ENDIAN
#endif
```

The program compiled and uploaded successfully to the wemos d1.

But the luck ended there. When I tried to start the VM, I couldn't get it. **Bytecode compilation problems started**:
```
line:1
err:6
1     print( "Hello World!\n" );for( i=0; i<10; i++ )        {                                print( i );              }                            print("\n");              
     ^
compile error [ERR_unexpected_token]
```
After removing the line `print( "Hello World!\n" );` the error appeared again:
```
line:1
err:6
1     for( i=0; i<10; i++ )        {                                print( i );              }                            print("\n");              
     ^
compile error [ERR_unexpected_token]
```

So, I assume that you use Arduino and do not work with the SDK from Espressif and I can only assume that you would know what the problem is?

I'll just add that I uncommented the line `#define WRENCH_SPIFFS_FILE_IO` and everything else is in default settings

jingoro2112, 2024-01-14:
thank you for the report and information on this tool chain! I am looking into it right now

jingoro2112, 2024-01-15:
Okay I downloaded the esp8366 toolchain (https://github.com/espressif/ESP8266_RTOS_SDK?tab=readme-ov-file) and reproduced your issue with the undefined endien-ness. 

I fixed up the code to compile correctly and as far as I can tell everything should work, but I don't have an esp8266 to test on. Could you please re-try your tests?

And how are you compiling the hello world app that is failing so immediately?

colesnicov, 2024-01-16:
Hello. The bytecode has compiled. But it doesn't work!

For the test I used the code from the file '[examples](https://github.com/jingoro2112/wrench/blob/main/examples/basic.cpp)':

```
void print( WRState* w, const WRValue* argv, const int argn, WRValue& retVal, void* usr )
{
	char buf[1024];
	for( int i=0; i<argn; ++i )
	{
		printf( "%s", argv[i].asString(buf, 1024) );
	}
}
static const char* wrenchCode =
"print( \"Hello World!\\n\" );"
"for( i=0; i<10; i++ )        "
"{                            "
"    print( i );              "
"}                            "
"print(\"\\n\");              ";

int XWr::DaemonStart()
{

	WRState* w = wr_newState(); // create the state

	wr_registerFunction( w, "print", print ); // bind a function

	unsigned char* outBytes; // compiled code is alloc'ed
	int outLen;

	int err = wr_compile( wrenchCode, strlen(wrenchCode), &outBytes, &outLen ); // compile it
	if ( err == 0 )
	{
		ESP_LOGE(LOG_TAG, "script has '%d' bytes!", outLen);
		wr_run( w, outBytes, outLen ); // load and run the code!
		delete[] outBytes; // clean up
	}

	wr_destroyState( w );
 ...
```

but it falls:
```
E (2320) XWr_daemon.cpp : 137: script has '71' bytes!
Guru Meditation Error: Core  0 panic'ed (LoadStoreAlignment). Exception was unhandled.
Core 0 register dump:
PC      : 0x4028355d  PS      : 0x00000030  A0      : 0x40283554  A1      : 0x3fff9f20  
0x4028355d: wr_newContext(WRState*, unsigned char const*, int) at wrench/src/wrench.cpp:12084

0x40283554: wr_newContext(WRState*, unsigned char const*, int) at wrench/src/wrench.cpp:12083

A2      : 0xbe712e96  A3      : 0x4010ac3f  A4      : 0xbe712e96  A5      : 0x4010a614  
A6      : 0x00000047  A7      : 0x00000042  A8      : 0x00000008  A9      : 0x00000090  
A10     : 0x402baa84  A11     : 0x00000045  A12     : 0x4010a574  A13     : 0x4010a614  
A14     : 0x3ffea8bc  A15     : 0x4010abfc  SAR     : 0x0000001d  EXCCAUSE: 0x00000009  

Backtrace: 0x4028355d:0x3fff9f20 0x4028747e:0x3fff9f50 0x402473cd:0x3fff9f60 0x4024b84e:0x3fff9f80 0x4024b534:0x3fff9fb0 
0x4028355d: wr_newContext(WRState*, unsigned char const*, int) at wrench/src/wrench.cpp:12084

0x4028747e: wr_run(WRState*, unsigned char const*, int) at wrench/src/wrench.cpp:12127

0x402473cd: xwr::service::XWr::DaemonStart() at XWr_daemon.cpp:139 (discriminator 1)

```

jingoro2112, 2024-01-16:
it seems to be crashing at the first time it tries to read the data in,
which suggests to me the heap is corrupt somehow. do you have any luck
running the pre-compiled example for example:

const int basic_bytecodeSize=53;
const unsigned char basic_bytecode[]=
{
0x00, 0x01, 0x04, 0x0D, 0x00, 0x48, 0x65, 0x6C, 0x6C, 0x6F, 0x20, 0x57,
0x6F, 0x72, 0x6C, 0x64, // 16
0x21, 0x0A, 0x06, 0x01, 0x88, 0x8A, 0x37, 0x16, 0xCE, 0x00, 0x00, 0xAE,
0x0A, 0x6F, 0x00, 0x0F, // 32
0x09, 0x1F, 0x00, 0x06, 0x01, 0x88, 0x8A, 0x37, 0x16, 0x8F, 0x00, 0x36,
0xEF, 0x09, 0x02, 0xEF, // 48
0x13, 0x99, 0x3E, 0xEC, 0xB6, // 53
};

I have checked this on all platforms I can and it works. I do have an esp
C6 I will run this on when I get it mounted :)

jingoro2112, 2024-01-16:
I just checked and the esp8266 is little endian, can you verify that my fix
(the defined I added in wrench.h) are properly identifying it as such?

jingoro2112, 2024-01-16:
Okay busted out an old 8266 and indeed it crashes even with compiled code.
looking into it now.

jingoro2112, 2024-01-16:
Got it! 

This is something I fixed a long time ago and forgot to not break again :(

the problem is reading *(int32_t*)ptr on unaligned memory locations. PC architecture doesn't mind this, including the PowerPC Big-Endian machine I test on, but esp32 very much bus-faults if you do it.

Fixed and checked in, should be good to go now!

closed 2024-01-16 by jingoro2112

colesnicov, 2024-01-18:
No. Is do not work.

jingoro2112, 2024-01-18:
Could you please make sure you get the latest? I found a definite issue
which I have fixed, I tested it on an actual esp8266 to confirm. 3.2.5
should have it