jingoro2112/wrench #20: Returning an array doesn't work as it used to

glebnovodran, 2024-01-15:
@jingoro2112 , could you please tell if this construction is supposed to work still?
https://github.com/glebnovodran/roam_bench/blob/9b90df43ec7af6ae93d649d31dfb550ca6a5d52b/src/acts/acts.w#L50

```
function SOME_STATE_FUNC() {
    //...
    newState = "some string";
    newStateDuration = 0.0;
    resetCollision = false;
   // ...
    res[] = {newState, newStateDuration, resetCollision};
    return  res;
}
``` 
It used to work with v3.0.0, but has stopped to work with the recent changes. The WRValue returned by wr_callFunction or by wr_returnValueFromLastCall is not recognized as an array anymore.

jingoro2112, 2024-01-15:
Yes it should, thank you for pointing this out I am fixing it right now.

glebnovodran, 2024-01-15:
Thank you for the prompt respnse, Curt.
By the way,  in the light of the future possible code changes, what do you see as a better way to obtain a result value from the function call?
wr_returnValueFromLastCall was in and out of the header file, if I am not mistaken, so I am not sure what is better to rely on?

jingoro2112, 2024-01-16:
This has been fixed, and your example added as part of the test suite. Sorry for the confusion. In all cases the return value is available directly from the wr_callFuncton(...) call:

```
function arrayCheck()
{
	newState = "some string";
	newStateDuration = 0.0;
	resetCollision = false;

	res[] = { newState, newStateDuration, resetCollision };

	return  res;
}
```
Might be accessed:
```
WRValue* V = wr_callFunction( context, "arrayCheck" );
if ( V )
{
	assert( V->isWrenchArray() );
	char someString[256];
	assert( WRstr(V->indexArray(context, 0, false)->asString(someString)) == "some string" );
	assert( V->indexArray(context, 1, false)->asFloat() == 0.0f );
	assert( V->indexArray(context, 2, false)->asInt() == 0);
	assert( !V->indexArray(context, 3, false)  );
	assert( V->indexArray(context, 3, true)->asInt() == 0  );
}
```

closed 2024-01-16 by jingoro2112

glebnovodran, 2024-01-16:
Thank you, Curt.