jingoro2112/wrench #22: Feature request: nested hash maps declaration (collections)

ohordiichuk, 2024-05-31:
Hello,

while it seems the VM is already supporting nesting the collections by storing them in different variables (as far as I understand from C++ code it works because variables could be stored as references):

```
// it works
var a = {"test1": 1, "test2": 4};
var b = {"test2" : 2};
var c = {"test4": 54};

b.test3 = a;
b.test3.test4 = c;
print(b.test3.test4.test4);
```

it's not possible to declare nested hash maps with 1 variable:

```
// it doesn't work (feature request)
var a = {"test1": {"test2" : 1}};
```

And also it's not possible to assign another collection to a key of existing collection:

```
// it doesn't work (feature request)
var a = {"test1" : 1};
a.test2 = {"test3" : 2};
```

jingoro2112, 2024-06-05:
hmm... yeah I can do this the first is easy but the second will be more of a challenge.

MaxTymchii, 2024-06-07:
Hi there @jingoro2112 ,

I want to achieve the next functionality,  but it seems there is no such a strait forward solution for that:

```
var table = {};
table["a"] = 100;
table["b"] = 1200;
table["c"] = 10;
```

Can you give me some advice on how to do that?

jingoro2112, 2024-06-07:
I am actually working on the hash/array parsing right now. This does work
but you need to declare table as a hash, not an array:

var table = {:};  // note ':' making this a hash table

table["a"] = 100;
table["b"] = 1200;
table["c"] = 10;

println( table["a"], " ",table["b"], " ",table["c"], " " );

produces the output:

./wrench r test.c100 1200 10

Without the colon you declared 'table' as an array. I definitely need
to make this more clear in the future, any suggestions are welcome!

closed 2024-06-07 by jingoro2112

MaxTymchii, 2024-06-07:
Thx! Now it's more clear. Adding such examples to your website will help to avoid such issues in the future. 

Another thing that I am working on now is testing scripts. 
The import feature is really helpful on it, but it's a lot of work that could be done in that field for me.

MaxTymchii, 2024-06-07:
One more off-topic thing. What IDE do you use for your development?

jingoro2112, 2024-06-07:
I totally agree that documentation and examples could be better. I'm sort
of hoping as wrench gets more use I'll visit more and more parts of it for
polish. Keep in mind as much as I take pride in fixing bugs quickly and
having clean code, it's just *me* and this is not my dayjob :)

For code-editing I use epsilon (https://www.lugaru.com/), but for general
development and debugging I use Visual Studio, currently 2022.

-Curt

jingoro2112, 2024-06-09:
This has been completed in version 5.0.1

jingoro2112, 2024-06-09:
Just replying in thread for anyone interested.. the current release 5.0.1
has all feature-requests (nested declaration etc) completed.

-Curt

ohordiichuk, 2024-06-09:
Curt, thank you for all your efforts! We will try everything out soon together with Max.

We have also successfully compiled wrench in Emscripten and will try to embed this library into our backend code. Later we can share details if that will be interesting for anybody. 

We also tried to create a syntax highlighting plugin for VSCode. It's very basic currently and not everything is working. But if we will manage to have something working good, we will share it.