jingoro2112/wrench #24: Feature request: native runtime errors

ohordiichuk, 2024-06-02:
It would be great if native function calls can stop `wr_run` execution if it's needed. For example, in case of bad parameters supplied by script or some fatal unrecoverable error that must stop further script execution. Since the library already has WRError, it should be possible to easily extend it for 3rd parties:

```c++
enum WRError {
  ...
WR_ERR_user = 10000, // some big value to make some space for future library's errors
}

void wr_setLastError(WRContext* c, WRError error) {
     c->w->err = error; // not sure if "w" can be accessed like that
}
```

And then in wr_callFunction in all places where `ccb` is called, e.g.:

```c++
CASE(CallFunctionByHash):
{
	args = READ_8_FROM_PC(pc++);

	// initialize a return value of 'zero'
	register0 = stackTop;
	register0->p = 0;
	register0->p2 = INIT_AS_INT;

	if ( ! ((register1 = w->globalRegistry.getAsRawValueHashTable(READ_32_FROM_PC(pc)))->ccb) )
	{
		w->err = WR_ERR_function_hash_signature_not_found;
		return 0;
	}

	register1->ccb( context, stackTop - args, args, *stackTop, register1->usr );
      
        if (w->err) {
               return 0;
        }

        ....
```

P.S.: Sorry, if I'm opening too many issues :) I just want to share our feedback using this library and make it better. I think I also can help with implementation if needed.

jingoro2112, 2024-06-02:
Not at all! I would absolutely love help on the implementation. Also any
feedback/amendments on the documentation would be most welcome, it's hard
to keep up. Sort of a one-programmer operation here :P

ohordiichuk, 2024-06-02:
Okay, great. I'll try to do this feature this week.

jingoro2112, 2024-06-03:
oops sorry too late I already implemented it :P

Also putting finishing touches on import, new turned out to be a pain
because the namespace table is linked into the imported code, so I had to
change the bytecode format a bit and push to v5.0.0

ohordiichuk, 2024-06-03:
Oh, no problem. Great you have already done it :)

jingoro2112, 2024-06-04:
okay so this is a PRE release version that includes import functionality.
it is NOT backwards compatible to 4.*

it includes wr_import(), sys::importCompile(), and sys::importByteCode()

Also there is sys::halt( err ) which takes an err argument between 100-255
as "user" level errors and will return immediately with that error.

I have some test code inside wrench_cli.cpp for the import testing but have
not added it to the unit tests yet, JUST got this working and am anxious
for you to try it out.

-Curt

jingoro2112, 2024-06-04:
OH.. in order for this to work you have to use the new "export" keyword on
structs that you want to export. this is not necessary for functions, they
just work, but 'export' ensures the namespace hash table is generated and
embedded in the bytecode, normally it only does this if it detects that you
'new'ed it somewhere. 'export' forces it to always be generated.

ohordiichuk, 2024-06-04:
Good job! We are happy to test it. But it seems you forgot to push the code into GitHub :)

jingoro2112, 2024-06-05:
okay current release implements this as sys::halt()

closed 2024-06-05 by jingoro2112