jingoro2112/wrench #28: Buggy hash table.

MaxTymchii, 2024-06-09:
I try to make assert for hash table and here what I get.

```
function assertHashTableEqual(actual, expected) {

var message = "";   
str::sprintf(message, "Hash Tables length: actual[%d], expected:[%d]", actual._count, expected._count);
println(message);
  if (actual._count != expected._count) {
    return false;
  }

    for (var key, var value : actual) {
        str::sprintf(message, "Key: %s value %d vs expected value %d ", key, actual[key], expected[key]);
        println(message);
        if (actual[key] != expected[key]) {
          return false;
        }
    }

    return true;
}   

var hashOne = {"a" : 1, "b" : 2, "c" : 3};
var hashTwo = {"a" : 1, "b" : 2, "c" : 3};

var hastableResult1 = assertHashTableEqual(hashOne, hashOne);
println("Equal: ");
println(hastableResult1);

var hashThree = {"a" : 1, "b" : 2, "c" : 3};
var hashFour = {"a" : 1, "b" : 2, "c" : 4};

var hastableResult2 = assertHashTableEqual(hashThree, hashFour);
println("Not Equal C:3 != C:4: ");
println(hastableResult2);

var hashFive = {"a" : 1, "b" : 2, "c" : 3};
var hashSix = {"a" : 1, "b" : 2, "c" : 3, "d" : 4};
var hastableResult3 = assertHashTableEqual(hashFive, hashSix);
println("Not Equal: length: 3 != 4:");
println(hastableResult3);

```

Results:

```
Hash Tables length: actual[10], expected:[10]
Key: a value 1 vs expected value 1 
Key: b value 2 vs expected value 2 
Key: c value 3 vs expected value 3 
Equal: 
1
Hash Tables length: actual[10], expected:[10]
Key: a value 1 vs expected value 1 
Key: b value 2 vs expected value 2 
Key: c value 3 vs expected value 4 
Not Equal C:3 != C:4: 
0
Hash Tables length: actual[10], expected:[10]
Key: a value 1 vs expected value 1 
Key: b value 2 vs expected value 2 
Key: c value 3 vs expected value 3 
Not Equal: length: 3 != 4:
1
```

length of each of the tables is not 10 they are 3 or 4!

Also, you can find another buggy thing.
If you remove from 
`for (var key, var value : actual) {`
`var value` it gonna be a strange behavior with a print.

MaxTymchii, 2024-06-09:
UPDATE:
Arrays are working properly with `_count`.

```
function assertEqualArray(actual, expected) {
    var message = "";
    str::sprintf(message, "Arrays length: actual[%d], expected:[%d]", actual._count, expected._count);
    println(message);
    if (actual._count != expected._count) {
        return false;
    } 

    for (var i = 0; i < actual._count; i++) {
        str::sprintf(message, "Index: %d, actual: %d, expected: %d", i, actual[i], expected[i]);
        println(message);
        if (actual[i] != expected[i]) {
            return false;
        }
    }
    return true;
}

```

```
Arrays length: actual[4], expected:[4]
Index: 0, actual: 1, expected: 1
Index: 1, actual: 2, expected: 2
Index: 2, actual: 3, expected: 3
Index: 3, actual: 4, expected: 4
PASS: [1, 2, 3, 4] and [1, 2, 3, 4] Arrays are equal
Arrays length: actual[4], expected:[4]
Index: 0, actual: 1, expected: 1
Index: 1, actual: 2, expected: 2
Index: 2, actual: 3, expected: 3
Index: 3, actual: 4, expected: 5
FAIL: [1, 2, 3, 4] and [1, 2, 3, 5] Arrays are NOT equal
Arrays length: actual[4], expected:[5]
FAIL: [1, 2, 3, 4] and [1, 2, 3, 4, 5] Arrays are NOT equal
```

jingoro2112, 2024-06-11:
most of these issues are fixed with the exception of hash size, which is a bit of a hassle.

I am leaving this open until I have a solution to that

jingoro2112, 2024-06-12:
hash size now works.

closed 2024-06-12 by jingoro2112