jingoro2112/wrench #30: Buggy text concat.

MaxTymchii, 2024-06-10:
While I was playing with texts I got next results:

```
var wrenchTest = io::readFile( "wrench_test_framework.w" );
sys::importCompile( wrenchTest );

var head = "Hello, playground";
var tail = " Goodbye";

var head1 = "Hello ";
var tail1 = ", World!";

println("Outside concat");
head = head + tail;
assertEqual( head, "Hello, playground Goodbye", "head = head + tail");

head1 += tail1;
assertEqual( head1, "Hello , World!", "head1 += tail1");

 var head2 = "New Hello ";
 var tail2 = "World!";
  function concat(tail) {
     println("Inside concat");
     head2 += tail;
     assertEqual( head2, "New Hello World!", "head2 += tail");
 }
concat("World!"); // Function is not called
concat(tail2); // Function is called

var head3 = "Super Hello ";
var tail3 = "Super World!";

function concater(tails) {
    println("Inside concater");
    head3 = head3 + tails;
    assertEqual( head3, "Super Hello Super World!", "head3 = head3 + tails");
}

concater(tail3); 

var head4 = "Super news ";
var tail4 = "Super star!";

function concater2(tail) {
    println("Inside concater");
    head4 += tail;
    assertEqual( head4, "Super news Super star!", "head4 += tail");
}

concater2(tail4);

```

Here are my results:

```
Outside concat
PASS: head = head + tail Actual: Hello, playground Goodbye
PASS: head1 += tail1 Actual: Hello , World!
Inside concat
PASS: head2 += tail Actual: New Hello World!
Inside concat
PASS: head2 += tail Actual: New Hello World!
Inside concater
FAIL: head3 = head3 + tails Actual: Super World!Super Hello , Expectation: Super Hello Super World!
Inside concater
FAIL: head4 += tail Actual: Super news , Expectation: Super news Super star!
```

You can check failed parts with strange concatenation behavior.

MaxTymchii, 2024-06-10:
UPDATE:

Not working properly
```
var head3 = "Super Hello ";
var tail3 = "Super World!";

function concater(tails) {
    println("Inside concater");
    head3 = head3 + tails;
    assertEqual( head3, "Super Hello Super World!", "head3 = head3 + tails");
}

concater(tail3); 
```

result:
```
Inside concater
FAIL: head3 = head3 + tails Actual: Super World!Super Hello , Expectation: Super Hello Super World!
```

Working hack:

```

var head5 = "It works ";
var tail5 = "Super works!";

function concater3(tail) {
    println("Inside concater 3");
    var insideHead = head5;
    insideHead = insideHead +  tail;
    assertEqual( insideHead, "It works Super works!", "insideHead += tail");
    head5 = insideHead;
    assertEqual( head5, "It works Super works!", "head5 = insideHead");
}

concater3(tail5);
```

Result:
```
Inside concater 3
PASS: insideHead += tail Actual: It works Super works!
PASS: head5 = insideHead Actual: It works Super works!
```

jingoro2112, 2024-06-10:
on it.

jingoro2112, 2024-06-14:
fixed in 6.0.1

closed 2024-06-14 by jingoro2112

MaxTymchii, 2024-06-14:
BIG THANKS!