jingoro2112/wrench #31: Missing WrenchValue::arraySize()

khevessy, 2024-06-11:
In the wrench.h in class WrenchValue, there is a method arraySize() but I cannot see the implementation anywhere, I think that it is missing.

MaxTymchii, 2024-06-11:
If you need a size you can try:
```
var array[] = {1, 2, 3, 4, 6};
println(array._count);
```

khevessy, 2024-06-11:
Thanks, but I am speaking about C++ method arraySize that is not implemented in the WrenchValue class:
```
class WrenchValue
{
public:

    WrenchValue( WRContext* context, const char* label ) : m_context(context), m_value( wr_getGlobalRef(context, label) ) {}
    WrenchValue( WRContext* context, WRValue* value ) : m_context(context), m_value(value) {}

    bool isValid() const { return m_value ? true : false; }

    // if the value is not the correct type it will be converted to
    // that type, preserving the value as best it can
    operator int32_t* () { return Int(); }
    int32_t* Int();

    operator float* () { return Float(); }
    float* Float();

    // this will convert it to an array if it isn't one
    WRValue& operator[] ( const int index ) { return *asArrayMember( index ); }
    WRValue* asArrayMember( const int index );
    int arraySize(); // returns -1 if this is not an array

private:
    WRContext* m_context;
    WRValue* m_value;
};
```

closed 2024-06-11 by jingoro2112