jingoro2112/wrench #34: Compilation error in gcc

ohordiichuk, 2024-06-14:
```
wrench/wrench/src/wrench.cpp:1973:54: error: 'strncasecmp' was not declared in this scope; did you mean 'strncmp'?
 1973 |     bool isMatchCase(const char* buf) const { return strncasecmp(buf, m_str, m_len) == 0; }
      |                                                      ^~~~~~~~~~~
      |                                                      strncmp
ninja: build stopped: subcommand failed.
```

compiler:
```
gcc version 12.2.0 (Zephyr SDK 0.16.5)
```

hot fix:
```
extern "C" {
#include <strings.h>
};
```

jingoro2112, 2024-06-14:
The purpose of strncasecmp is to compare two strings safely ignoring case.
Which is what isMatchcase() does in that string class. It serves no purpose
in wrench (it's from my own library) and I have removed it in 6.0.1

-Curt

jingoro2112, 2024-06-14:
fixed

closed 2024-06-14 by jingoro2112