jingoro2112/wrench #37: Can't debug if starting line is a function call

darksotmoon, 2024-06-25:
Yes, debugging might be work in progress, but it seems mostly working.

Works
```
function test_fn()
{
  print( "In function\n" );
}
print("start");
test_fn();
print("end");
```

Does not work
```
function test_fn()
{
  print( "In function\n" );
}
// print("start");
test_fn();
print("end");
```

If wr_callFunction the error is WR_ERR_cannot_call_function_context_yielded but I am unsure why the byte code has a yield command without disassembling it.

jingoro2112, 2024-06-25:
broke is broke :)

I am working on it on time available, there are a couple of really bad bugs
with debug/stepping right now. I folded the debug functionality into yield,
which didn't exist when first laid down the debugger, and there is some
pain there. I will try to release a working debugger ASAP, even if it's not
fully functional, it should at least do the basic stepping/variable display.

jingoro2112, 2024-06-30:
I think you will find that these cases both now work properly, with step into and over working :)

I am continuing to refine the debugger, its still very much alpha but at least works!

closed 2024-06-30 by jingoro2112

darksotmoon, 2024-06-30:
I believe that my error was from my own memory corrupt when capturing the printf output.  Apologies about that.

But I went ahead and made a disassembler and the output is better with your changes.  Before the changes `CallFunctionByIndex` would have two junk opcodes following it.  

```c
function test_fn()
{
  print("In function\n");
}
// print("start");
test_fn();
print("end");
```

```asm
; disassembly

; size 0x6F 111

; hash 'print' = 0x16378a88

; header
0x0000 0x00 number globals = 0
0x0001 0x01 number units = 2
0x0002 0x03 compiler flags = 0x03
0x0003 0x0000 unit namespace offset
0x0005 0x004C unit function offset
0x0007 0x745399A2 unit hash
0x000B 0x00 unit args
0x000C 0x00 frame space
0x000D 0x01 frame adjust
0x000E 0xFD468B0E source hash
0x0012 0x0017 symbol size = m_symbolBlockLen 23
0x0014 0x02 unit count = 2
0x0016 unit 0
0x0016 0x00 locals = 0
0x0017 0x00 args = 0
0x0018 0x3A 0x3A 0x67 0x6C 0x6F 0x62 0x61 0x6C 0x00 '::global' hash 0x46A75FDE
0x0021 unit 1
0x0021 0x00 locals = 0
0x0022 0x00 args = 0
0x0023 0x74 0x65 0x73 0x74 0x5F 0x66 0x6E 0x00 'test_fn' hash 0x745399A2
0x002B end of header

0x002B unit 0 ::global
; test_fn();
0x002B 0xF5 DebugInfo 4006 line number 6
0x002E 0xF5 DebugInfo 0001 WRD_FunctionCall
0x0031 0x07 CallFunctionByIndex args 0 index 0 pop 3
; print("end");
0x0037 0xF5 DebugInfo 4007 line number 7
0x003A 0x04 LiteralString size 3 'end'
0x0040 0xF5 DebugInfo 0E00 WRD_FunctionCall
0x0043 0x06 CallFunctionByHashAndPop args 0x01 hash 0x16378A88
0x0049 0x02 LiteralZero 
0x004A 0xEF GlobalStop 
0x004B 0x14 Stop 

0x004C unit 1 test_fn
;   print("In function\n");
0x004C 0xF5 DebugInfo 4003 line number 3
0x004F 0x04 LiteralString size 12 'In function\n'
0x005E 0xF5 DebugInfo 0E00 WRD_FunctionCall
0x0061 0x06 CallFunctionByHashAndPop args 0x01 hash 0x16378A88
; }
0x0067 0xF5 DebugInfo 4004 line number 4
0x006A 0x12 ReturnZero 

0x006B 0xE704BBC2 final hash
```

jingoro2112, 2024-06-30:
you have a disassembler? I've almost written one a dozen times but never
found the time. If you want me to add it to the repo send it on over! I'll
give you full credit though I'll probably modify it :)

darksotmoon, 2024-06-30:
It's a quick hack that supports only the opcodes that I used in the sample source.  And I never bothered to learn the underlying method of the VM.    And there is the ugly `get_source_line`, and mostly but not quite C.

```c++
#include "wrench.h"
#include <vector>

////////////////////////////////////////////////////////////////////////////////

#define MAX_PAYLOAD_BYTES 4

typedef struct {
	uint8_t opcode;
	int payload[MAX_PAYLOAD_BYTES];
} opcode_t;

typedef enum { PL_NONE, PL_ARG, PL_INDEX, PL_HASH, PL_POP } payload_t;

static const opcode_t opcode_s[] =
{
	{ O_CallFunctionByHash, { PL_ARG, PL_HASH }},
	{ O_CallFunctionByHashAndPop, { PL_ARG, PL_HASH }},
	{ O_CallFunctionByIndex, { PL_ARG, PL_INDEX, PL_POP }},
	{ O_PushIndexFunctionReturnValue, { PL_INDEX }},
};
static const int opcode_count = sizeof(opcode_s) / sizeof(opcode_t);

////////////////////////////////////////////////////////////////////////////////

static int offset(uint8_t * bytecode, uint8_t * code)
{
	return code - bytecode;
}

static const char * get_source_line(const char * source, int line)
{
	int l = 0;
	const char * r = source;
	const char * st = source;
	while((l < line) && *r)
	{
		st = r;
		if(*st == '\r')
		      st++;
		while(*r)
		{
			if(*r == '\n')
			{
				l++;
				r++;
				break;
			}
			 r++;
		}
	}
	return st;
}

void wr_disassemble(unsigned char * bytecode, int bytecode_len, const char * source, const char * filename)
{
	FILE * out = fopen(filename, "w+");
	if(out != NULL)
	{
		uint8_t * code = bytecode;
		std::vector<std::string> unit_names;

		fprintf(out, "; disassembly\n");
		fprintf(out, "\n");
		fprintf(out, "; size 0x%X %d\n", bytecode_len, bytecode_len);
		fprintf(out, "\n");
		fprintf(out, "; hash 'print' = 0x%.08x\n", wr_hashStr("print"));
[... 213 lines cut ...]

				fprintf(out, "0x%.04X 0x%.02X %s%s\n", offset(bytecode, start), opcode, (opcode <= O_DebugInfo) ? c_opcodeName[opcode] : "!!!", desc);
			}
		}

		fprintf(out, "\n");
		fprintf(out, "0x%.04X 0x%.08X final hash\n", offset(bytecode, code), *(uint32_t *)code);

		fclose(out);

	}

	#if 0
	std::string dump_name = filename;
	dump_name += ".bin";
	FILE * dump = fopen(dump_name.c_str(), "w+");
	if(dump != NULL)
	{
		uint8_t * code = bytecode;
		fwrite(bytecode, bytecode_len, 1, dump);
		fclose(dump);
	}
	#endif
}
```