jingoro2112/wrench #43: WRD_DebugOut breaks code debugging

darksotmoon, 2024-07-03:
Test script
```
debug::print("debug out");
str::printf("end");
```

The first single step will stop on line 1 as expected.  Client sends `WRD_RequestStepInto`, server executes script and sends back `WRD_Ok`.

Then the next single step will fail.   What happens is that the client `WRDebugClientInterfacePrivate::trapRunOutput` will send `WRD_RequestStepInto`, then the server executes the code, which involves sending back `WRD_DebugOut` then `WRD_Ok`.

`WRDebugClientInterfacePrivate::trapRunOutput` will receive `WRD_DebugOut` in the loop and then send the `WRD_RequestStepInto` again.  The server executes this and sends back `WRD_Ok` so the client ends up with three `WRD_Ok` replies.  Not sure where the third `WRD_Ok` comes from.

The problem is that `WRDebugClientInterfacePrivate::transmit` is really `transmit_receive` so it can't receive two packets ( `WRD_DebugOut` then `WRD_Ok`) without transmitting again.

I renamed transmit to transmit_receive and then added transmit and receive functions.  Then `WRDebugClientInterfacePrivate::trapRunOutput` looks like:

```
if(transmit( WrenchPacketScoped(WrenchPacket::alloc(packet)) ))
{
	bool done = false;
	while(!done)
	{
		WrenchPacket *reply = receive();
		if(reply)
		{
			if( reply->_type == WRD_DebugOut )
			{
				printf( "%s", (char*)reply->payload() );
			}
			else
				done = true;
			g_free( reply );
		}
	}
}
```

jingoro2112, 2024-07-03:
okay I'll look into this as soon as I can, but I am off for the week :)

jingoro2112, 2024-10-09:
just noting I have not forgotten, but debug remains buggy and until I need it is not going to get any love.

darksotmoon, 2024-10-10:
That's ok by me.  The non-debug functionality is working just fine.