jingoro2112/wrench #44: compiling on arduino mega

alex4300, 2024-09-11:
First, I'm really amazed about this project!
Wanted to do something like that for ages, but I guess wouldn't got that far.

To my Issue, I tried to adapt the basic.cpp  example to Arduino, but it seems too freeze the arduino completely when compiling, even Serial.print output gets truncated to the serial monitor.
 
Running precompiled bytecode works.

I'm using Arduino IDE 2.3.2 and an Arduino Mega 2560.

Here is the relevant example-code:

```

const char* wrenchCode = 
"print( \"Hello World!\\n\" );"
"for( var i=0; i<10; i++ )    "
"{                            "
"    print( i );              "
"}                            "
"print(\"\\n\");              ";

void setup()
{
	Serial.begin( 115200 );

	delay( 2000 ); // wait for link to come up for sample

	Serial.print( "\r\n" );

	WRState* w = wr_newState(); // create the state with 20 stack entries

	wr_registerFunction( w, "print", print ); // bind a function

	//wr_run( w, basic_bytecode, basic_bytecodeSize ); // load and run the code!
	unsigned char* outBytes; // compiled code is alloc'ed
  
	int outLen;
        Serial.println(wrenchCode);
        Serial.println(strlen(wrenchCode));
        delay( 200 ); // Need some delay here, or wr_compile freezes output before data sent.
  
	int err = wr_compile(wrenchCode, strlen(wrenchCode), &outBytes, &outLen); // compile it
	if ( err == 0 )
	{
            Serial.print("compiled: ");
            Serial.println(outLen);

		wr_run( w, outBytes, outLen, true ); // load and run the code!
	}
  else {
    Serial.print("error: ");
    Serial.print(err);
  }

	wr_destroyState( w );
}

```

alex4300, 2024-09-11:
Here is the full Example, it outputs just:

```
READY
C

```

So the output from Serial.print( "COMPILE\r\n" ); is already truncated.

```
#include "wrench.h"

const int basic_bytecodeSize=54;
const unsigned char basic_bytecode[]=
{
	0x01, 0x00, 0x00, 0x04, 0x0D, 0x00, 0x48, 0x65, 0x6C, 0x6C, 0x6F, 0x20, 0x57, 0x6F, 0x72, 0x6C, // 16
	0x64, 0x21, 0x0A, 0x06, 0x01, 0x88, 0x8A, 0x37, 0x16, 0xCF, 0x00, 0x00, 0xAF, 0x0A, 0x70, 0x00, // 32
	0x0F, 0x09, 0x20, 0x00, 0x06, 0x01, 0x88, 0x8A, 0x37, 0x16, 0x90, 0x00, 0x37, 0xEF, 0x09, 0x02, // 48
	0xEF, 0x14, 0x94, 0xF8, 0x8F, 0x5A, // 54
};

void print( WRContext* c, const WRValue* argv, const int argn, WRValue& retVal, void* usr )
{
	char buf[512];
	for( int i=0; i<argn; ++i )
	{
		Serial.print( argv[i].asString(buf, 512) );
	}
}

const char* wrenchCode = 
"print( \"Hello World!\\n\" );"
"for( var i=0; i<10; i++ )    "
"{                            "
"    print( i );              "
"}                            "
"print(\"\\n\");              ";

void setup() {
  Serial.begin( 115200 );

	delay( 2000 ); // wait for link to come up for sample

	Serial.print( "READY\r\n" );

	WRState* w = wr_newState(); // create the state with 20 stack entries
  unsigned char* outBytes; // compiled code is alloc'ed
  int outLen;

  wr_registerFunction( w, "print", print ); // bind a function
 
  Serial.print( "COMPILE\r\n" );
  int err = wr_compile( wrenchCode, strlen(wrenchCode), &outBytes, &outLen ); // FREEZES here
  if ( err == 0 )  {
      wr_run( w, outBytes, outLen, true ); // load and run the code!
  }
//	wr_run(w, basic_bytecode, basic_bytecodeSize ); // BUT this works...

}

void loop() {

}

```

jingoro2112, 2024-09-12:
The wrench VM is very very lightweight but the compiler? not so much :)

Small embedded chips like the ATMega will blow chunks trying to run it.
They can execute the pre-compiled code just fine (thats all that bytecode
is) but cannot compile, sorry.

-Curt

alex4300, 2024-09-12:
First of all, thanks for the reply!

I already figured it out that this might be the case ;-)
Trying to compile already stops at the assert( sizeof(int) == 4 ); line in wr_compile(), so I'll guess compiling at least depends on 32bit integers.

What do you think, are the hurdles generally too big to overcome to get that running on a mega or could it be possible to get it running to a degree with some hacks?

Would love to get this running!

jingoro2112, 2024-10-09:
The odds are near zero, sorry. Getting the compiler to work on teeny embedded processors is not a design goal of this project :(

closed 2024-10-09 by jingoro2112