jingoro2112/wrench #45: Interpreter Seg Faults instead of returning an error for invalid scripts

ander-theo, 2024-10-08:
Hello, thanks for all your hard work on wrench.
First of all, my use case is a C++ program running on a microcontroller that accepts and runs wrench scripts from an end user.  Since they are created by the end user, these scripts could very well be gibberish or invalid code.  I need to be able to run any script sent by an end user without it crashing the program, as if the wrench code were a type of sandbox for the user to try things out in without being able to bring down the system by accident.

While testing, I have come across two cases so far where a certain script, when run, will seg fault and crash.
1. Divide by zero eg. "function run(){10/0;}"
2. Misuse of runtime variable eg. "function run(values){ values+=10;}" where values at runtime is passed a hashtable.  The same issue occurs for -=, *=, /=
This list probably isn't exhaustive.

What I need is for wrench to protect against all seg faults or other terminating errors so that no script can crash the program.  Ideally, any invalid script would return a runtime (or compiler) error or follow some defined non-crashing behavior.

ander-theo, 2024-10-08:
For more context, I did some debugging and here is where example 2 seg faults:

[image]

jingoro2112, 2024-10-09:
Okay looking into this

jingoro2112, 2024-10-09:
So even though at that point in the code it is guaranteed to be an extended (EX) type, it was NOT guaranteed to actually be an array, case in point it was a hash table :P

So that has been amended, along with a complete set of traps for divide-by-zero.

closed 2024-10-09 by jingoro2112

ander-theo, 2024-10-14:
Thank you!  Would you be able to update the platformio registry to 6.0.5 when you get a chance?

jingoro2112, 2024-10-15:
oh, yeah done.