jingoro2112/wrench #48: Function wr_freeString is not defined

timarnoldev, 2024-12-19:
The function wr_freeString is mentioned in the documentation but not available in the code.

I'm currently tackling a memory leak.
Everytime I call print using this function some 100ish bytes of memory leak:

```c++
inline void print(WRContext* c, const WRValue* argv, const int argn, WRValue& retVal, void* usr)
    {
        char buf[128];
        for (int i = 0; i < argn; ++i)
        {
            printf("%s", argv[i].asString(buf, 1024));
        }
    }
```

```wrench
    print("Hello World\n");
```

As I'm executing the print function in a loop, i quickly run out of memory.
Do I have to manually have to free the string passed from wrench?
Am I missing something here?

jingoro2112, 2024-12-19:
most definitely should not be happening I'll fix this immediately. I run
the entire unit test suite through valgrind with zero tolerance so I don't
know whats going on here.

-Curt

jingoro2112, 2024-12-19:
> Okay this is my fault for making the usage unclear.
>
> Minor problem: you are creating a buffer of 128 characters but telling asString()
> that 1024 are available, not a problem in your example but will cause an
> overrun if you call 'print' with a string larger than 128.
>
> Second problem is that wr_run() returns a malloc'ed context. This context
> is tracked in the state, and will be destroyed when you call
> wr_destroyState(), but not until then! You can destroy it yourself with
> wr_destroyContext() but that is not clear with wr_run's usage.
>
> I provide a macro right below that:
>
> // macro for automatically freeing the WRContext
> #define wr_runOnce( w, block, blockSize ) wr_destroyContext(
> wr_run((w),(block),(blockSize)) )
>
> but this is easy to miss :) so I've changed the function to now be this:
>
> // w:          state (see wr_newState)
> // block:      location of bytecode
> // blockSize:  number of bytes in the block
> // context:    pointer to store the context that will be allocated.
> //             NOTE: If not supplied the context will be automatically
> //             cleaned up, and wr_run will return null.
> // takeOwnership: assume 'block' was allocated with wr_malloc(), it
> //                will be freed with  wr_free()
> WRContext* wr_run( WRState* w, const unsigned char* block, const int
> blockSize, WRContext** context =0, bool takeOwnership =false );
>
> which should make it more obvious what's going on and how to deal with it.
> I am open to suggestions on usage though, any ideas? The context needs to
> be preserved in many cases for callbacks to work after the code has been
> run.
>
> -Curt

tried using it and immediately hated it. So I changed it to this:

// w:          state (see wr_newState)
// block:      location of bytecode
// blockSize:  number of bytes in the block
// destroyContext: automatically destroy the resulting context, by
//                 default wr_run will return the allocated context
//                 this context can be used for callbacks or destroyed
//                 with wr_destroyContext(), but will be garbage
//                 collected automatically when wr_destroyState()
//                 is called on w
// takeOwnership: assume 'block' was allocated with wr_malloc(), it
//                will be freed with  wr_free()
WRContext* wr_run( WRState* w,
                   const unsigned char* block,
                   const int blockSize,
                   const bool takeOwnershipOfBlock =false,
                   const bool destroyContext =false );

which I like way better.

timarnoldev, 2024-12-19:
What I forgot to mention is the fact that I run the whole script once and then call a function over and over using wr_callFunction. This is where the memory leaks. The whole script looks something like this:

```wrench
function fun() {
      print("Hello World\n");
}
```

Is there an option to clear the context after each execution of a function?

jingoro2112, 2024-12-19:
> memory leaks. The whole script looks more like this:

that shouldn't leak, there must be a gc call missing. I'll check on that.

jingoro2112, 2024-12-19:
This is my test program, it does not exhibit a leak, I have traced it a bit
and don't see what you might be encountering but I certainly believe
*something* is wrong so please advise.

inline void LEAKprint(WRContext* c, const WRValue* argv, const int argn,
WRValue& retVal, void* usr)
{
    char buf[1024];
    for (int i = 0; i < argn; ++i)
    {
        printf( "%s", argv[i].asString(buf, 1024) );
    }
}

const char* LEAKwrenchCode = "function leak() { print( \"Hello World!\\n\"
); }";

void LEAKtest()
{
    WRState* w = wr_newState();

    wr_registerFunction( w, "print", LEAKprint );

    unsigned char* outBytes = 0;
    int outLen;

    wr_compile( LEAKwrenchCode, (int)strlen(LEAKwrenchCode), &outBytes,
&outLen );

    WRContext* ctx = wr_run( w, outBytes, outLen, true );
    for(;;)
    {
        wr_callFunction( ctx, "leak" );
    }

    wr_destroyState( w );
}

timarnoldev, 2024-12-20:
After further investigations I found out that after roughly 20kb of consumed memory a huge block is released again. So calling it a memory leak is probably the wrong term. My problem is, that sometimes I don't have 20kb available causing the esp8266 to run oom.

Is there a way to manually invoke gc or directly mark the string as deleteable?

jingoro2112, 2024-12-21:
found it. The gc "hint" was not accounting for some of the data from my new
rework, so the 4k minimum allocation before GC was called was not being
properly honored. I have added a test to specifically check for this in the
future so all allocations are properly accounted for. Of course I didn't
notice on the PC I usually test with. Sorry. I need to run the new
version through some proper tests before releas it but this will be patched
in the next few days.

To add the fix your current version the offending line is:

 allocatedMemoryHint += ret->init( size, type, init );

which should read:

 allocatedMemoryHint += ret->init( size, type, init ) + sizeof(WRGCObject);

Thank you for finding this! Any other issues you encounter please let me
know I will be very responsive.

-Curt

timarnoldev, 2024-12-22:
I just tried the fix and it works like intended. I'm looking forward to the next release.
Thanks for your help.

closed 2024-12-22 by timarnoldev