jingoro2112/wrench #49: Concatenating a string with a number results in a string of random numbers

timarnoldev, 2024-12-22:
If have something like this is wrench:

```wrench
var points = 0;
print("Your points: "+points);
```

The c implementation of "print":

```c++
inline void print(WRContext* c, const WRValue* argv, const int argn, WRValue& retVal, void* usr)
    {
        for (int i = 0; i < argn; ++i)
        {
            char buf[128];
            printf("%s\n", argv[i].asString(buf, 128));
        }
    }
```

The output in the console is for example:

1635393679

Only logging a string or a number works fine.
Are there any limitations in wrench regarding concatenation?

jingoro2112, 2024-12-22:
This is most likely a type coercion oversight and should definitely work
the expected way

jingoro2112, 2024-12-22:
I will look into this as part of thenext update, but holidays are slowing
me down 😞

timarnoldev, 2025-01-07:
I checked again with the new released version

**print("Hello " + "World");**

results in

**Hello Hello**

with

```c++
inline void print(WRContext* c, const WRValue* argv, const int argn, WRValue& retVal, void* usr)
{
    char buf[1024];
    for (int i = 0; i < argn; ++i)
    {
        printf("%s\n", argv[i].asString(buf, 1024));
    }
}
```

When I try to run it on the esp8266 it throws an exception:

<details><summary>Details</summary>
<p>

No poison after block at: 0x3fff53a7, actual data: 0x20 0x48 0x65 0x14

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic umm_malloc.cpp:751 umm_malloc_core

>>>stack>>>

ctx: cont
sp: 3ffffa70 end: 3fffffd0 offset: 0010
3ffffa80:  40100aa8 3ffe8794 0000071f 40100ddc
3ffffa90:  00000d89 3ffe8794 3fff32b8 40100b72
3ffffaa0:  40100779 3ffe8794 000002fa 00000006
3ffffab0:  4020734e 3fff3294 00000020 401010fc
3ffffac0:  6c6c6548 3fff3254 00000006 40101167
3ffffad0:  00000012 3fff4ffc 00000020 401010fc
3ffffae0:  00000005 3fff4ffc 00000018 40100619
3ffffaf0:  00000000 3fff3294 00000006 4020734e
3ffffb00:  00000006 3fff45ec 3fff1f5c 40100619
3ffffb10:  00000005 3fff4ffc 3fff3294 4020755a
3ffffb20:  00000000 40208e4b 40208fe7 00000001
3ffffb30:  3fff50d8 00000000 3fff45f8 00000006
3ffffb40:  00000000 3fff50c8 3ffea617 40208c98
3ffffb50:  40208e13 40208c48 40208c52 40208c67
3ffffb60:  40208c72 40208e4b 40208fe7 40209106
3ffffb70:  40209198 402091d5 40209280 40209319
3ffffb80:  402094c7 40209407 402094fa 40209549
3ffffb90:  402095fe 4020967a 4020968a 40209692
3ffffba0:  4020972f 4020975c 40209798 402097a0
3ffffbb0:  402097df 40209813 4020988d 4020984a
3ffffbc0:  40209855 40209929 40209952 40209997
3ffffbd0:  402099b5 402099d3 402099ef 40209a0e
3ffffbe0:  40209a2d 4020a213 4020a21c 4020a225
3ffffbf0:  4020a22e 4020a237 4020a240 4020a286
3ffffc00:  4020a28f 4020a298 4020a2a1 4020a2aa
3ffffc10:  4020a2b3 4020a2bc 4020a2c5 4020a2ce
3ffffc20:  4020a2d7 40209a49 40209a8b 40209aac
3ffffc30:  40209ac4 40209b1d 4020a75c 4020a762
3ffffc40:  4020a768 4020a771 4020a77a 4020a78c
3ffffc50:  4020a783 4020a7e7 4020aa67 4020aa44
3ffffc60:  4020aa79 4020aa3b 4020aa70 4020aa4d
3ffffc70:  4020aa93 4020ab0f 4020aa9c 4020ab18
3ffffc80:  4020aaa5 4020ab21 4020a849 4020a891
3ffffc90:  4020a855 4020a8a5 4020a879 4020a8e1
3ffffca0:  4020a885 4020a8f5 4020a861 4020a8b9
3ffffcb0:  4020a86d 4020a8cd 4020abb7 4020ac06
3ffffcc0:  4020aba3 4020abed 4020ab91 4020abdb
3ffffcd0:  4020ab9a 4020abe4 4020abc0 4020ac0f
3ffffce0:  4020abc9 4020ac18 4020ac55 4020ac9f
3ffffcf0:  4020ac43 4020ac8b 4020ac31 4020ac79
3ffffd00:  4020ac3a 4020ac82 4020ac5e 4020aca8
3ffffd10:  4020ac67 4020acb1 4020ad03 4020acc5
3ffffd20:  4020acfa 4020acce 4020acf1 4020acd7
3ffffd30:  4020adb2 4020ad7a 4020ada9 4020ad71
3ffffd40:  4020ada0 4020ad83 4020ad59 4020ad1b
3ffffd50:  4020ad50 4020ad24 4020ad47 4020ad2d
3ffffd60:  4020ae10 4020add8 4020ae07 4020adcf
3ffffd70:  4020adfe 4020ade1 4020a191 4020a1d3
3ffffd80:  4020a541 4020a557 4020a55d 4020a565
3ffffd90:  4020a56d 4020a579 4020a58d 4020a59b
3ffffda0:  4020a34c 4020a3fc 40209ce6 40209cf4
3ffffdb0:  40209d48 4020a2ea 4020a2f3 4020a30e
3ffffdc0:  4020a2fc 4020a305 4020a317 4020a320
3ffffdd0:  4020a329 4020a332 4020a33b 4020a357
3ffffde0:  4020a360 4020a37b 4020a369 4020a372
3ffffdf0:  4020a384 4020a38d 4020a396 4020a39f
3ffffe00:  4020a3a8 40209b6b 40209bae 40209d92
3ffffe10:  40209db0 40209c05 40209bec 40209c5c
3ffffe20:  40209cdb 40209c53 40209cb7 4020a448
3ffffe30:  4020a451 4020a45a 4020a463 4020a4c5
3ffffe40:  4020a4ce 4020a4d7 4020a4e0 4020a92d
3ffffe50:  4020a936 4020a912 4020a909 4020a91b
3ffffe60:  4020a924 4020a9cb 4020a9d4 4020a9b0
3ffffe70:  4020a9a7 4020a9b9 4020a9c2 4020a5a7
3ffffe80:  4020a615 4020a67b 4020a68c 40209dd7
3ffffe90:  40209dfc 40209e2a 40209e33 40209e58
3ffffea0:  40209e86 40209e92 40209ea4 4020a69d
3ffffeb0:  4020a6b5 4020a6d2 4020a6ea 4020a6f3
3ffffec0:  4020a6fc 4020a705 4020a70e 4020a717
3ffffed0:  4020a72f 4020a738 4020a741 4020a74a
3ffffee0:  4020a753 40209ee0 40209f28 40209f73
3ffffef0:  40209f7c 40209f93 40209fa7 40209fb0
3fffff00:  40209fbe 4020a09f 4020a12b 40209727
3fffff10:  402096b3 402096ed 40208d08 40208d75
3fffff20:  40208ddb 40208e3e 3fffff60 3fff12a4
3fffff30:  3fff50c8 3fff505c 3fff4ffc 3fff45ec
3fffff40:  3fff50c0 00000000 00000000 3ffea616
3fffff50:  00000000 3fff50d8 00855555 4020ae6c
3fffff60:  00000000 00000000 00000000 3fff12a4
3fffff70:  43110000 00000000 3fff1f5c 4020644e
3fffff80:  00000000 00000c10 051eb851 000df285
3fffff90:  3fffdad0 3fff12a4 402153be 3fff12a4
3fffffa0:  3fffdad0 00000000 3fff1278 4020653b
3fffffb0:  00000000 00000000 00000001 4021554c
3fffffc0:  feefeffe feefeffe 3fffdab0 401014c1
<<<stack<<<

0x40100aa8 in get_poisoned at umm_malloc.cpp:?
0x40100ddc in umm_malloc_core at umm_malloc.cpp:?
0x40100b72 in check_poison_block at umm_malloc.cpp:?
0x40100779 in puts at ??:?
0x4020734e in WRGCObject::init(unsigned int, WRGCObjectType, bool) at ??:?
0x401010fc in umm_malloc at ??:?
0x40101167 in umm_poison_malloc at ??:?
0x401010fc in umm_malloc at ??:?
0x40100619 in malloc at ??:?
0x4020734e in WRGCObject::init(unsigned int, WRGCObjectType, bool) at ??:?
0x40100619 in malloc at ??:?
0x4020755a in WRContext::getSVA(int, WRGCObjectType, bool) at ??:?
0x40208e4b in wr_callFunction(WRContext*, WRFunction*, WRValue const*, int) at ??:?

</p>
</details>

jingoro2112, 2025-01-07:
Thanks, crashes are always a priority. I will be fixing this today and have
a fix for it asap.

-Curt

jingoro2112, 2025-01-07:
So I have the main problem fixed and added a bunch more test code:

var h2 = "Hello ";
var w2 = "World";
if ( (h2 += w2) != "Hello World" ) println( "hw_err6__" );

var h = "Hello " + "World";
if ( h != "Hello World" ) println( "hw_err0" );
var h1 = "Hello ";
var w1 = "World";
if ( h1 + w1 != "Hello World" ) println( "hw_err1" );
if ( h1 + "World" != "Hello World" ) println( "hw_err2" );
if ( "Hello " + w1 != "Hello World" ) println( "hw_err3" );
if ( "Hello " + "World" != "Hello World" ) println( "hw_err4" );
if ( h1 != "Hello " ) println( "hw_err5" );
if ( w1 != "World" ) println( "hw_err5" );

var h2 = "Hello ";
var w2 = "World";
if ( (h2 += w2) != "Hello World" ) println( "hw_err6" );
h2 = "Hello ";
if ( (h2 += "World") != "Hello World" ) println( "hw_err7" );

h2 = "Hello ";
var h3 = "foo";
if ( (h3 = h2 + w2) != "Hello World" ) println( "hw_err8" );
h3 = "foo";
if ( (h3 = h2 + "World") != "Hello World" ) println( "hw_err9" );
h3 = "foo";
if ( h3 = "Hello " + "World" != "Hello World" ) println( "hw_err10" );

var a1 = { "Hello " };
var a2 = { "w":"World" };
if ( a1[0] + "World" != "Hello World" ) println( "hw_err11" );
if ( a1[0] + a2["w"] != "Hello World" ) println( "hw_err12" );
if ( "Hello " + a2["w"] != "Hello World" ) println( "hw_err13" );

Running through lots more testing now, will release in the next few hours.

-Curt

timarnoldev, 2025-01-08:
Thanks for the fix. Concatenation of two strings works now.
Adding a flag to auto convert to string, when concatenating a string a number would be the best option in my opinion.
I agree that this is not that expressive, but teaching the add operator is much simpler than introducing string formating.
It is also the same way as it works in Java, if you start with a string and add a number it is automatically converter to a string

jingoro2112, 2025-01-08:
yet another reason Java hurts my soul.

Okay I think I can work that in. will be part of the next release, probably
within a week depending on my dayjob schedule and how much snow we get (I'm
in Atlanta)

-Curt

jingoro2112, 2025-01-08:
Okay 6.0.13 is up and properly executes the following test code:

var s = "string";
if ( s + 123 != "string123" ) println( "st_err0" );
if ( "string" + 123 != "string123" ) println( "st_err1" );
if ( 123 + "string" != "123string" ) println( "st_err2" );
if ( 123 + s != "123string" ) println( "st_err3" );
if ( s != "string" ) println( "st_err4" );
if ( (s += 123) != "string123" ) println ( "st_err5" );
var i = 123;
if ( (i += "string") != "123string" ) println ( "st_err6" );
if ( (123 + "string" + 34 + "hi") != "123string34hi" ) println ( "st_err7"
);

jingoro2112, 2025-01-08:
> Oh I didn't realized that there are more libs than documented on the
> website. I will definitely check them out. Thanks for sharing!

it's a one-man show here :P

I document when I can, should really do more I know.

timarnoldev, 2025-01-08:
No worries!
Wrench already helps me so much.
Thanks for your awesome work.

jingoro2112, 2025-01-08:
> oops never implemented those calls.. thats probably why they are not in
> the documentation.. I'll look a bit deeper when I have the chance.
>
> And you are helping me way more than I am helping you. Exercising the
> features really helps iron out issues.

(just stop me if I am spamming you)

I remember now I started implementing that and then added the basics inline
and forgot about this lib. I am updating it now and will provide the
following:

 wr_registerLibraryFunction( w, "array::clear", wr_listCount );       // (
array )
wr_registerLibraryFunction( w, "array::count", wr_listCount );       // (
array )
wr_registerLibraryFunction( w, "array::remove", wr_arrayRemove );    // (
array, where [,count] )
wr_registerLibraryFunction( w, "array::insert", wr_arrayInsert );    // (
array, items, where )
wr_registerLibraryFunction( w, "array::tuncate", wr_arrayTruncate ); // (
array, newSize )

wr_registerLibraryFunction( w, "hash::clear", wr_hashClear );   // ( hash,
item, key )
wr_registerLibraryFunction( w, "hash::count", wr_hashClear );   // ( hash,
item, key )
wr_registerLibraryFunction( w, "hash::add", wr_hashAdd );       // ( hash,
item, key )
wr_registerLibraryFunction( w, "hash::remove", wr_hashRemove ); // ( hash,
key )
wr_registerLibraryFunction( w, "hash::peek", wr_hashPeek );     // ( hash,
key )
wr_registerLibraryFunction( w, "hash::pop", wr_hashPop );       // ( hash,
key )

wr_registerLibraryFunction( w, "list::clear", wr_arrayClear );        // (
list )
wr_registerLibraryFunction( w, "list::count", wr_arrayCount );        // (
list )
wr_registerLibraryFunction( w, "list::peek", wr_arrayPeek );          // (
list )
wr_registerLibraryFunction( w, "list::pop", wr_arrayPop );            // (
list )
wr_registerLibraryFunction( w, "list::pop_front", wr_arrayPop );      // (
list )
wr_registerLibraryFunction( w, "list::pop_back", wr_arrayPopBack );   // (
list )
wr_registerLibraryFunction( w, "list::push", wr_arrayPush );          // (
list, item )
wr_registerLibraryFunction( w, "list::push_front", wr_arrayPush );    // (
list, item )
wr_registerLibraryFunction( w, "list::push_back", wr_arrayPushBack ); // (
list )

wr_registerLibraryFunction( w, "queue::clear", wr_arrayClear );   // ( q )
wr_registerLibraryFunction( w, "queue::count", wr_arrayCount );   // ( q )
wr_registerLibraryFunction( w, "queue::push", wr_arrayPush );     // ( q,
item )
wr_registerLibraryFunction( w, "queue::pop", wr_arrayPopBack );   // ( q )
wr_registerLibraryFunction( w, "queue::peek", wr_arrayPeekBack ); // ( q )

wr_registerLibraryFunction( w, "stack::clear", wr_arrayClear ); // ( stack )
wr_registerLibraryFunction( w, "stack::count", wr_arrayCount ); // ( stack )
wr_registerLibraryFunction( w, "stack::push", wr_arrayPush );   // ( stack,
item )
wr_registerLibraryFunction( w, "stack::pop", wr_arrayPop );     // ( stack )
wr_registerLibraryFunction( w, "stack::peek", wr_arrayPeek );   // ( stack )

Any comments are welcome. bear in mind in order to keep things simple
wrench can handle arrays and hashes but not linked-lists natively, so the
list/q/stack functionality is going to me faked with an array, but should
be good for learning at least. Wrench was *not* designed for that. It was
designed to be very fast, very small, very light and easy to use.

-Curt

timarnoldev, 2025-01-09:
That sounds really good.
Looking forward to the release.

jingoro2112, 2025-01-10:
okay it's up as 6.0.14 with the documentation amended and these functions
added:

array::clear( array, [size] );          // clear/create an array the
specified size (default 0)
                                        // returns: the array
array::count( array );                  // returns: elements
array::remove( array, where, [count] ); // remove count items (default 1)
                                        // from where (items shifted down)
                                        // returns: the array
array::insert( array, where, [count] ); // add count blank items (default 1)
                                        // at where (items shifted up)
                                        // returns: the array
array::truncate( array, size ); // chop off this array at the given size
                                // returns: the array

hash::clear( hash );          // clear/create a hash table
                              // returns: the table
hash::count( hash );          // number of items in the hash table
                              // return: number
hash::add( hash, item, key ); // add item as location key,
                              // clobbering/creating as necessary
                              // returns: the item
hash::remove( hash, key );    // remove this key from the hash table
                              // returns: 1 if the key was found,
                              //          otherwise 0
hash::exists( hash, key );    // return 1 if this key is found otherwise 0

list::clear( list );            // clear/create a list
                                // returns: the list
list::count( list );            // returns: elements
list::peek( list );             // return the first element of the list
without
                                // changing it
list::pop( list );              // return and remove the first element of
the list
list::pop_front( list );        // return and remove the first element of
the list
list::pop_back( list );         // return and remove the last element of
the list
list::push( list, item );       // push item to front of the list
list::push_front( list, item ); // push item to front of list
list::push_back( list, item );  // push item to back(end) of list

queue::clear( queue );      // clear/create a queue (filo)
queue::count( queue );      // returns: elements
queue::push( queue, item ); // push item onto the queue
queue::pop( queue );        // pop an item off the queue
queue::peek( queue );       // return the first element of the queue without
                            // changing it

stack::clear( stack );      // clear/create a stack (fifo)
stack::count( stack );      // return: elements
stack::push( stack, item ); // push item onto the stack
stack::pop( stack );        // pop an item off the stack
stack::peek( stack );       // return the first element of the stack without
                            // changing it

jingoro2112, 2025-01-10:
oops.. gimmee another day on that new lib, I accidentally checked it in
with most of the test code disabled and valgrind is complaining a bit on
some of the edges. I'm sorting it out now.

jingoro2112, 2025-01-10:
good to go, please let me know what you think of the API and any
additions/changes you might suggest

jingoro2112, 2025-01-10:
By the way thanks for requesting string + non string concatenation work. I
find myself writing test code like this now so easily:

println( "1:" + i[1] + " 2:" + i[2] + " 3:" + i[3] + " 4:" + i[4] + " 5:" +
i[5] + " 6:" + i[6] );

You've also got me thinking, adding a linked-list element to wrench
shouldn't be hard as it is a very fundamental type, I just don't have the
bits left over to describe it in my WRValue struct! I'm staring at it now
though to see if I can add it. I'd rather the list/queue/stack work as
linked lists like they are supposed to.

Oh! And you wanted context highlighting. I should make a pass at learning
to do that for visual code at least.

Also going to add x:y( ... ) as syntactic sugar for y( x, ... ) the way lua
does. I like that for "free" object-oriented style programming.

Sorry to sound scattered, I am a professional programmer but it's also my
hobby.

-Curt

jingoro2112, 2025-01-15:
fixed

closed 2025-01-15 by jingoro2112