jingoro2112/wrench #50: Unexpected compiler output

timarnoldev, 2025-01-14:
When compiling the following code wrench fails with this error:

```number(1, 4, list::count(snake_x)-1, 0xFF0000);```

```
line:47
err: WR_ERR_var_not_seen_before_label[22]
47        }
          ^
```

If I add spaces before and after the "-" than the error is gone:

```number(1, 4, list::count(snake_x) - 1, 0xFF0000);```

jingoro2112, 2025-01-14:
> ouch. I will fix that parsing error immediately.
> If you need more information, hit me up

Yeah the problem is it's parsing it as "(list::count(snake_x)) (-1)"

Rather than the "list() - 1" that is clearly intended.

*sigh* This will be a fun one to fix. sorry you had to stumble into it, and
thank you for reporting it.

jingoro2112, 2025-01-15:
fixed in 6.0.16

closed 2025-01-15 by jingoro2112