jingoro2112/wrench #53: Direct struct allocation in C++

khevessy, 2025-02-19:
Thank you for your work. I tested #51 and it works as expected. I have a follow-up: now that structure allocated from Wrench can be used directly, would you consider making it possible to somehow create Wrench structure in C++, so that when it is passed as an argument, it can be directly copied without changing its type? Basically I would like to mirror the structure in the outside code. What I mean is this:

Wrench:
```
struct Frame
{
    var Channel;
    var Id;
    var Length;
}
var frame = new Frame ();

/* This is called from C++ */
function RxEvent(externalFrame)
{
    frame = externalFrame;
    frame.Length = 5;
    /* callback back to C++ */
    callback(frame);
}
```
`externalFrame` is created using `wr_makeContainer()` and then `wr_makeInt()` and  `wr_addValueToContainer()` (as outlined in the documentation). I notice that when this is done, type of the `frame` that is passed  back to C++ is changed from struct to hashtable. I know that the elements can be accessed directly, so when I want to preserve the structure data type, I have to do this:
```
frame.Channel = externalFrame.Channel;
frame.Id = externalFrame.Id;
frame.Length = 5;
```
This could be done, but for larger structures, it seemed like an inefficient syntax.

Do you think my use case is valid? Sorry for the convoluted explanation.

jingoro2112, 2025-02-20:
Okay I think I can meet you halfway on this, all it took was a sleepless
night and a good shower. Gimmee a day or so.

jingoro2112, 2025-02-22:
6.0.18 implements:

// create an instance of the named struct (as if it were new'ed inside
// wrench) if the name is not a valid struct in the given context the
// return value is null.
// This value is automatically cleaned up when the context is destroyed
WRValue* wr_instanceStruct( WRValue* val, WRContext* context, const char* name, const WRValue* argv =0, const int argn =0 );

which lets you create and pass structs :)

closed 2025-02-22 by jingoro2112

khevessy, 2025-02-25:
It works as expected, thank you Curt.